dune3d/dune3d #159: Proposal change dimensional arrow shape

Peta-T, 2025-03-13:
I have tested different values of this constants in file renderer.cpp:  

static const float constraint_arrow_scale = .005;
static const float constraint_arrow_aspect = 6.;

arrows now look much more like machinery dimensional arrow.
Thanks

carrotIndustries, 2025-03-13:
Have a screenshot of how this looks like?

carrotIndustries, 2025-03-13:
Personal preference. Unless other users also consider this worth changing, I don't see a reason for different arrows shapes.

closed 2025-03-13 by carrotIndustries

Peta-T, 2025-03-13:
there are some standards of arrows for machinery, building or other design areas 

[image]

[image]

carrotIndustries, 2025-03-13:
I see, though with your setting the arrows are too tiny to my taste.