dune3d/dune3d #160: Proposal to change reference planes behavior as one reference cross with 3 selectable planes

Peta-T, 2025-03-13:
From my experience with other cad systems it is to much better to have one reference cross system  with separately selectable 3 plaanes, 3 vectors and center point.

[image]

You can have more crosses with possibility to be independent or with relative link to other cross.

carrotIndustries, 2025-03-13:
What can be done with this approach that can't be done with the workplanes that we have right now?

Peta-T, 2025-03-13:
for example if you will need shape oriented like this and after you will need perpendicular plane it is not easy to find right angles for perpendicular plane
 
[image]

Peta-T, 2025-03-13:
If you will do some big cast part with more than hundred solid partbodies(extrude etc...) you will need more crosses also for possibility  to make changes on this big part. In some companies is modeling standard to make one cross for each solid partbody.

carrotIndustries, 2025-03-13:
What you describe as 'cross' is more or less already available as a workplane entities.

What could be added is a constraint that constrains a workplane's rotation relative to another one.

Peta-T, 2025-03-13:
OK

Peta-T, 2025-03-13:
for next example if you can imagine that in automotive design 3d model of car have one common axis cross for all components of car.
Its good solution for big assemblies an for colaboration you do not need make constraints only append to zero position to main assembly, but
also other tools for production of those parts from car have also this one common axis cross which is often about 4m long distance from part to this axis cross. You will need another cross for designing those parts.