dune3d/dune3d #172: Selection on macOS seems to be broken

deviant, 2025-03-31:
Hi! I just tried building Dune 3D on macOS. Had no issues getting it to compile, and it launches without issues. Interacting with the application mostly seems to work, I can open menus and use tools. Almost all mouse/trackpad functionality is present; I can pan/tilt without issues. However, I cannot seem to select nodes at all!

Clicking on nodes/lines/measurements doesn't do anything, and whilst I can summon a selection rectangle by dragging, it doesn't seem to catch anything. I also noticed that sometimes the selection rectangle will hang around even after letting go of the mouse/trackpad, so perhaps the application isn't registering click-end events inside the viewport properly? Clicking on everything else (even on the tool buttons, which are over the viewport area) works without issues, as far as I can tell.

Example of ghost selection rectangle:

[image]

I know that [the documentation](https://github.com/dune3d/dune3d-docs/blob/18c25cdf3206a5c19a797d11131b2ffa6da3578c/build-macos.rst) says that selection is broken, but I found a [couple](https://github.com/dune3d/dune3d/pull/25#issuecomment-1875853026) [of](https://github.com/dune3d/dune3d/pull/45) [issues](https://github.com/dune3d/dune3d/issues/57) referencing this problem which have since been closed, and I couldn't find an issue dedicated to this problem in general, so I thought I'd create a new one.

I am on the latest version of macOS at the time of writing, Sequoia 15.3.2 (24D81), on an Intel MacBook. All listed dependencies were at their latest version, and I tried to make sure that I didn't have anything running that could have interfered with pointer input.

jue89, 2025-03-31:
Same here. I own a quite old MacBook from 2016. macOS Monterey is the latest support macOS version.

viernullvier, 2025-03-31:
Strange, haven't seen any selection issues for quite some time.

1. Which input device are you using? Selection works for me as expected both with the internal Macbook touchpad and an external Magic Touchpad, but I haven't tried a normal mouse yet.
2. Can you check which specific GTK version you're building against? This wouldn't be the first time that an upstream GTK update would break something on macOS.

viernullvier, 2025-03-31:
Are you able to reproduce the issue on the latest CI-built Mac binaries?

https://github.com/dune3d/dune3d/actions/runs/14173740876

This one was still built against GTK 4.18.2, by the way.

deviant, 2025-03-31:
> Are you able to reproduce the issue on the latest CI-built Mac binaries?
> 
> [dune3d/dune3d/actions/runs/14173740876](https://github.com/dune3d/dune3d/actions/runs/14173740876)
> 
> This one was still built against GTK 4.18.2, by the way.

I am able to reproduce the issue. (on the Intel build, I sadly don't have an M-series machine to test with.)

carrotIndustries, 2025-03-31:
To debug this, see the comments in https://github.com/dune3d/dune3d/issues/57 for enabling dumping the pick buffer.

deviant, 2025-03-31:
Here's a sketch:

[image]

and the corresponding pick buffer:

[image]

The dump file is just filled with lines of zeroes repeating. Looks like it's not being filled whatsoever for some reason.

deviant, 2025-04-01:
I opened #175 to track the "ghost selection rectangle" problem, as it seems to be unrelated to this issue. I haven't gotten any closer to figuring out what's causing this, but I suspect it's due to part of the rendering pipeline failing (perhaps `glReadPixels` doesn't work?)

viernullvier, 2025-04-02:
I just checked back on my old Intel MacBook and confirmed my suspicion: This issue has been going on for a while and it's not even the first time I'm seeing it. I haven't touched the dune3d repo on the old machine for over a year, but it already showed the same problem. I'll do a `git bisect` to figure out where exactly things went wrong.

viernullvier, 2025-04-02:
Turns out that even the very first macOS builds were having this issue. There's something going on that's not supported on Intel-based Macs (or Intel and AMD GPUs) which works just fine on ARM-based Macs and on Linux and Windows.

Unfortunately I know next to nothing about OpenGL, but I'd suspect that the issue might be one of these:
- Pick buffer data doesn't get properly passed between/returned from the shaders due to platform-specific quirks
- Using multiple color attachments has some platform-specific quirks
- There are some platform-specific restrictions for blitting between framebuffers
- `glReadPixels` has some platform-specific restrictions

However, I've also found this: https://mcneel.myjetbrains.com/youtrack/issue/RH-73275 - others are struggling with empty framebuffers on Intel-based Macs as well. Rhino is closed-source, but the commit messages shared in the thread give a tiny hint about potential issues:

> Metal: Use a managed memory buffer instead of shared
> - Shared memory buffers are not supported on AMD and Intel GPUs.

and

> Metal: Use synchronizeResource on Metal with managed storage

However, these issues seem to be Metal-related, so I have no idea how they translate to OpenGL.

ruevs, 2025-04-02:
> Turns out that even the very first macOS builds were having this issue.

Indeed that is the case. https://github.com/dune3d/dune3d/pull/25#issuecomment-1811596329

I'm not on a Mac but have been following Dune3D almost from "day 1" :-)

viernullvier, 2025-04-05:
Folks, I've figured it out - you wouldn't believe how simple the fix is.

In #45, we've already established that higher MSAA settings don't work on macOS (leads to a hard crash), so we're limiting things to a sampling factor of 1x. All good, the app runs, even the selection mechanism works on arm64 devices.

However, a sampling factor of 1x doesn't seem to be directly equivalent to the corresponding non-multisampled OpenGL functions: According to the docs, it's actually a sampling factor of `0` that will exactly reproduce the behavior of the non-multisampled functions. On arm64, a sampling factor of `1` just happens to work by accident, but this is not the case on the Intel platform: Here, it still renders, but transferring the rendered scene to the pick buffer is broken. I can't tell where exactly it breaks, but at least there's now a trivial fix available - just defaulting to `0`.

So far I haven't noticed any visual differences between a sampling factor of `1` and `0` on arm64, but I encourage everyone to take a closer look at this once I've set up the PR.

PR #176 (2025-04-05): mac: fix broken selection on intel platform
Fixes #172.

There seems to be a platform-specific difference in treatment of the MSAA sampling factor. On arm64-based Macs, setting it to `1` works as expected, but it breaks rendering to the pick buffer on Intel-based Macs. Setting it to `0` ensures compatibility with both platforms.
closed 2025-04-05 by carrotIndustries