dune3d/dune3d #173: Hard Crash on macOS when trying to save or trying to open the settings

rafo, 2025-03-31:
Nice Project!

The app crashes every time when I try to save my work. It also crashes every time when I try to open the settings dialog. The "About" dialog works fine ;D

VERSION:
I used the macOS ARM build from the GitHub Action output: https://github.com/dune3d/dune3d/actions/runs/14088044578 after a fresh reboot.

SETUP:
OS Version: macOS 15.3.2 (24D81)
Model: MacBookPro17,1, BootROM 11881.81.4, proc 8:4:4 processors, 16 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online

```
Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libglib-2.0.0.dylib           	       0x1036583ac _g_log_abort + 40
1   libglib-2.0.0.dylib           	       0x103658740 g_log_structured_array + 264
2   libglib-2.0.0.dylib           	       0x1036581f8 g_log_default_handler + 204
3   libglib-2.0.0.dylib           	       0x103657dec g_logv + 828
4   libglib-2.0.0.dylib           	       0x103657a40 g_log + 28
5   libgio-2.0.0.dylib            	       0x10350a094 g_settings_set_property + 264
6   libgobject-2.0.0.dylib        	       0x102edb58c object_set_property + 196
7   libgobject-2.0.0.dylib        	       0x102edae20 g_object_new_internal + 288
8   libgobject-2.0.0.dylib        	       0x102edaae0 g_object_new_valist + 1104
9   libgobject-2.0.0.dylib        	       0x102eda410 g_object_new + 32
10  libgio-2.0.0.dylib            	       0x103506dc8 g_settings_new + 44
11  libgtk-4.1.dylib              	       0x1039ff0f4 gtk_color_chooser_widget_init + 380
12  libgobject-2.0.0.dylib        	       0x102eef0f4 g_type_create_instance + 404
13  libgobject-2.0.0.dylib        	       0x102edad44 g_object_new_internal + 68
14  libgobject-2.0.0.dylib        	       0x102eda5f4 g_object_new_with_properties + 460
15  libgtk-4.1.dylib              	       0x1039ec184 _gtk_builder_construct + 728
16  libgtk-4.1.dylib              	       0x1039f3e28 builder_construct + 40
17  libgtk-4.1.dylib              	       0x1039f2be8 end_element + 1260
18  libgtk-4.1.dylib              	       0x1039f18b0 proxy_end_element + 68
19  libglib-2.0.0.dylib           	       0x103654a38 emit_end_element + 152
20  libglib-2.0.0.dylib           	       0x103654410 g_markup_parse_context_parse + 3400
21  libgtk-4.1.dylib              	       0x1039f1650 _gtk_builder_parser_parse_buffer + 320
22  libgtk-4.1.dylib              	       0x1039ed544 gtk_builder_add_from_resource + 360
23  libgtkmm-4.0.0.dylib          	       0x1046b6e20 Gtk::Builder::add_from_resource(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&) + 56
24  dune3d-bin                    	       0x101191344 dune3d::CanvasPreferencesEditor::create(dune3d::Preferences&) + 72
25  dune3d-bin                    	       0x101174eac dune3d::PreferencesWindow::PreferencesWindow(dune3d::Preferences&) + 476
26  dune3d-bin                    	       0x100ea545c dune3d::Dune3DApplication::show_preferences_window(unsigned int) + 64
27  dune3d-bin                    	       0x100eaa38c dune3d::Dune3DApplication::on_startup()::$_1::operator()() const + 36
28  dune3d-bin                    	       0x100eaa35c decltype(std::declval<dune3d::Dune3DApplication::on_startup()::$_1&>()()) std::__1::__invoke[abi:fe190107]<dune3d::Dune3DApplication::on_startup()::$_1&>(dune3d::Dune3DApplication::on_startup()::$_1&) + 24
29  dune3d-bin                    	       0x100eaa338 std::__1::invoke_result<dune3d::Dune3DApplication::on_startup()::$_1&>::type std::__1::invoke[abi:fe190107]<dune3d::Dune3DApplication::on_startup()::$_1&>(dune3d::Dune3DApplication::on_startup()::$_1&) + 24
30  dune3d-bin                    	       0x100eaa314 decltype(auto) sigc::adaptor_functor<dune3d::Dune3DApplication::on_startup()::$_1>::operator()<>() const + 24
31  dune3d-bin                    	       0x100eaa2f0 sigc::internal::slot_call<dune3d::Dune3DApplication::on_startup()::$_1, void>::call_it(sigc::internal::slot_rep*) + 40
32  libgiomm-2.68.1.dylib         	       0x104258ed8 (anonymous namespace)::SimpleAction_signal_activate_connect_callback(_GSimpleAction*, _GVariant*, void*) + 116
33  libgobject-2.0.0.dylib        	       0x102ed4588 g_closure_invoke + 200
34  libgobject-2.0.0.dylib        	       0x102eebf48 signal_emit_unlocked_R + 1940
35  libgobject-2.0.0.dylib        	       0x102eeaec4 signal_emit_valist_unlocked + 1992
36  libgobject-2.0.0.dylib        	       0x102eea6d0 g_signal_emit_valist + 64
37  libgobject-2.0.0.dylib        	       0x102eeb23c g_signal_emit + 28
38  libgio-2.0.0.dylib            	       0x1034fa158 g_simple_action_activate + 172
39  libgiomm-2.68.1.dylib         	       0x1041e8d14 Gio::Action_Class::activate_vfunc_callback(_GAction*, _GVariant*) + 252
40  libgtk-4.1.dylib              	       0x103b78b28 gtk_action_muxer_activate_action + 152
41  libgtk-4.1.dylib              	       0x103b78b40 gtk_action_muxer_activate_action + 176
42  libgtk-4.1.dylib              	       0x103b78b40 gtk_action_muxer_activate_action + 176
43  libgtk-4.1.dylib              	       0x103b57da0 gtk_widget_activate_action_variant + 68
44  AppKit                        	       0x18cb6b454 -[NSApplication(NSResponder) sendAction:to:from:] + 460
45  AppKit                        	       0x18cc33b0c -[NSMenuItem _corePerformAction] + 372
46  AppKit                        	       0x18d35cc90 _NSMenuPerformActionWithHighlighting + 152
47  AppKit                        	       0x18cc6b4d4 -[NSMenu performActionForItemAtIndex:] + 200
48  AppKit                        	       0x18cc6b3f4 -[NSMenu _internalPerformActionForItemAtIndex:] + 76
49  AppKit                        	       0x18d3529d0 +[NSCocoaMenuImpl _performActionForMenuItem:] + 176
50  AppKit                        	       0x18d158e8c -[NSMenuTrackingSession _performPostTrackingDismissalActions] + 284
51  AppKit                        	       0x18d158b84 -[NSMenuTrackingSession startRunningMenuEventLoop:] + 1332
52  AppKit                        	       0x18d1585f4 -[NSMenuTrackingSession startMonitoringEvents:] + 256
53  AppKit                        	       0x18d217818 -[NSMenuBarTrackingSession _mouseDownEventHandler:] + 416
54  AppKit                        	       0x18d217638 -[NSMenuBarTrackingSession _handleMonitorEvent:] + 408
55  AppKit                        	       0x18d217328 __57-[NSMenuBarTrackingSession _addLocalEventMonitorIfNeeded]_block_invoke + 104
56  AppKit                        	       0x18cbc01e0 _NSSendEventToDequeuingObservers + 252
57  AppKit                        	       0x18d3311f0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 2172
58  libgtk-4.1.dylib              	       0x103cbe634 poll_func + 184
59  libglib-2.0.0.dylib           	       0x10365153c g_main_context_iterate_unlocked + 380
60  libglib-2.0.0.dylib           	       0x103651604 g_main_context_iteration + 60
61  libgio-2.0.0.dylib            	       0x1034efdc0 g_application_run + 540
62  dune3d-bin                    	       0x100ea2dcc main + 60
63  dyld                          	       0x1889ec274 start + 2840

```

viernullvier, 2025-03-31:
This sounds familiar - IIRC this can happen if some GTK-specific settings directories have not been defined or set up. Will look into it.

viernullvier, 2025-03-31:
Turns out that I haven't tested the bundle thoroughly enough - I can reproduce this on my local machine as well. Fix incoming in #174, @rafo please test the macOS app bundles once the CI workflows are finished.

rafo, 2025-03-31:
It works! I am testing the app for several minutes. No crash so far. Wonderful!

PR #174 (2025-03-31): mac: fix missing gsettings schema in bundle
Fixes #173.

The bundled app is unable to find the local gsettings schema definitions because they haven't been included in the bundle. This leads to a crash when opening the preferences (related to the color picker schema) or when saving a file (related to the file chooser schema).
closed 2025-03-31 by carrotIndustries