dune3d/dune3d #183: Exception when saving

deviant, 2025-04-24:
Got this little popup when pressing Cmd-S to save the document. Also happens when I press the "Save" button. Recurs after pressing "Ignore" and trying to save again.

[image]

deviant, 2025-04-24:
Trying to open "Save As" also produces the same behaviour.

carrotIndustries, 2025-04-25:
Since this seems to be reproducible: Start Dune 3D in gdb like so: `gdb build/dune3d`
In the GDB shell:
```
catch throw
run
````

When the exception is thrown, run `bt` to get a backtrace.

deviant, 2025-04-25:
Unfortunately I have not seen the bug recur after restarting Dune 3D. Would I be able to attach to the process after the fact with `gdb -p` and acquire a backtrace if I were to see it happen again in the future?

carrotIndustries, 2025-04-25:
When the dialog is there, it's already too late. If you can reproduce it with a running instance of Dune 3D, this approach will work.

I've heard that modern mac OS doesn't allow debugging other processes out of the box.

deviant, 2025-04-25:
Thanks for the info! I suppose I'll have to just always launch it like this for a while, and see if it shows up again. Sucks how restrictive macOS is by default these days.