dune3d/dune3d #184: Points do not appear when rendering to discrete GPU

deviant, 2025-04-24:
On a couple of occasions now, I've had it that points aren't appearing. This makes it impossible to select them, perform point constraints, etc.

[image]

When missing, points do not show up in the pick buffer either:

[image]

This appears to be dependent on whether I opened Dune while using my discrete GPU or just the integrated one:

[image] [image]

Using macOS on an Intel machine.

carrotIndustries, 2025-04-25:
Interesting. Constraint icons use the same icon shader as points and constraints are visible. Maybe try playing a bit with the icon shaders to figure out where it's going wrong.

> This appears to be dependent on whether I opened Dune while using my discrete GPU or just the integrated one:

So only one of then is broken?

deviant, 2025-04-25:
> Interesting. Constraint icons use the same icon shader as points and constraints are visible. Maybe try playing a bit with the icon shaders to figure out where it's going wrong.

Will do, thanks for the pointer!

> So only one of then is broken?

Correct. Points show up as expected on the integrated GPU, but are missing on the discrete one.

deviant, 2025-04-27:
I've not touched any shader code in _years_, and I never got particularly deep into it back then, so working my way through this has been pretty slow. But I've made some progress! If I disable `vec` being transformed and passed through in `icon-vertex.glsl`, we get points:

[image]

Can you explain what `vec` is for? I'm assuming it's a small offset from its source position, so that icons don't overlap lines/etc.

I'm wondering if the point icons are just being rendered off-screen or something, given this code seems to be doing offsetting and/or perspective shifting. Would love some suggestions for what to focus on next.

carrotIndustries, 2025-04-27:
That `vec` thing is there to rotate the constraint icons such that they always align with the entity they apply to:

[image]

Maybe the NAN doesn't quite find its way to the shader and it takes the `!isnan` branch even though the value is garbage.

deviant, 2025-04-28:
> Maybe the NAN doesn't quite find its way to the shader and it takes the `!isnan` branch even though the value is garbage.

I'm not quite sure what the reason is, but the result from `isnan()` is nonsense, yeah. Emulating with a self-comparison makes the bug go away, which implies that it's still being treated as a NaN in the shader? Almost certainly a driver bug of some sorts.

PR #189 (2025-04-28): workaround macOS AMD GPU driver bug
For some reason, `isnan()` was always returning false, even when being passed a NaN. Given that `isnan(x)` is essentially equivalent to testing if `x != x` (additionally suppressing exceptions from signalling NaNs, which we shouldn't run into here), the latter can just be substituted in instead.

Fixes #184.