dune3d/dune3d #186: File extension is duplicated (on macOS)

deviant, 2025-04-24:
No special steps are needed for this, it just happens whenever trying to save to a new file:

[image]

Also in export dialogs:

[image]

deviant, 2025-04-24:
Correction: this does not happen when saving a new file, but it does happen with the "Save As" dialogue. So, once the document has a name already, this will happen.

viernullvier, 2025-05-02:
Can't reproduce it here (macOS 14.1, ARM64, GTK 4.18.2). Does this behavior depend on the global Finder settings to show/hide file extensions?

deviant, 2025-05-02:
> Does this behavior depend on the global Finder settings to show/hide file extensions?

I was also wondering if that was related, but it doesn't seem to make a difference. Haven't tested to see if rebooting after changing that setting makes a difference. I have file extensions enabled everywhere, FWIW.