dune3d/dune3d #191: [Bug] Reference group behaves weirdly

viernullvier, 2025-05-02:
The reference group is supposed to be special, but the modification safeguards have a couple of gaps.

### Lines in 3D

**Steps to reproduce:**
1. Select Reference group
2. Draw a line in 3D
3. Set a distance constraint on the drawn line
4. Change the distance value
5. Try to delete the line

**Expected behavior:**
Option A:
 - Drawing and constraining the line should not be possible
Option B:
- The line length should be changed to the given distance value
- The line should be deleted

**Actual behavior:**
- Constraints can be entered, but the solver does not run and will not update the line
- The line can not be deleted (the constraint can be deleted though)

### Drawing in workplane

**Steps to reproduce:**
1. Select Reference group
2. Set one of the workplanes as active
3. Draw something (circle, line etc)
4. Apply a constraint to the drawn entities
5. Try to delete the drawn entities

**Expected behavior:**
Option A:
 - Drawing and constraining sketches should not be possible
Option B:
- The constraint should be applied, the entities' geometries should be modified
- The entities should be deleted

**Actual behavior:**
- Constraints can be entered, but the solver does not run and will not update the entities
- Entities can not be deleted (constraints can be deleted though)

### New workplanes

**Steps to reproduce:**
1. Select Reference group
2. Create a new workplane
3. Select both the new workplane and the ZX or YZ workplane, apply "Constrain same orientation"
4. Try to delete the workplane

**Expected behavior:**
Option A:
 - Creating and constraining the workplane should not be possible
Option B:
- The workplane should change orientation
- The workplane should be deleted

**Actual behavior:**
- The "same orientation" constraint appears, but the solver does not run and will not rotate the workplane
- The workplane can not be deleted (the constraint can be deleted though)

### Thoughts

It makes sense to treat the reference group as special - if anything, deleting the three base workplanes should not be possible as it will probably cause unexpected behavior. The most straightforward fix should probably just disable the creation of new entities as well, but I'm wondering if it would actually make sense to allow reference entities in the reference group? One example I'm thinking of is an imported DXF file as a base drawing that should be the reference for the rest of the model. In some cases, I've been ending up creating a second reference group anyway just to maintain a set of base geometries to use as a reference for later actual sketches.

carrotIndustries, 2025-05-03:
> In some cases, I've been ending up creating a second reference group anyway just to maintain a set of base geometries to use as a reference for later actual sketches.

That's the intended use model.

I think the best way to fix this issue is to disallow creating entities in the reference group.

carrotIndustries, 2025-05-04:
See the linked branch.

viernullvier, 2025-05-06:
Yup, the branch seems to work as expected.

commit 3b2273c3 (2025-05-06): disallow creating entities and constraints in the reference group

changelog: Bugfixes/Editor: disallow entities and constraints in reference group

fixes #191
closed 2025-05-06 by carrotIndustries