dune3d/dune3d #193: execution error of locally built dune3d

wsteffe, 2025-05-08:
The build finished without errors on my linux system (Ubuntu 20.24) but when I try to execute it I got following errors:

> ./dune3d
> realize
> Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
> 0(3) : error C7532: global type vec2 requires "#version 100" or later
> 0(4) : error C7532: global type vec3 requires "#version 100" or later
> 0(4) : error C5060: out can't be used with non-varying color_to_fragment
> 0(9) : error C7532: global variable gl_VertexID requires "#version 130" or later
> 0(9) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
> 0(10) : error C7532: global function pow requires "#version 100" or later
> 0(15) : error C7532: global variable gl_Position requires "#version 100" or later
> 0(15) : error C7532: global type vec4 requires "#version 100" or later
>  
> Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
> 0(3) : error C7532: global type vec3 requires "#version 100" or later
> 0(7) : error C5060: out can't be used with non-varying normal_to_fragment
> 0(8) : error C5060: out can't be used with non-varying color_to_fragment
> 0(9) : error C5060: out can't be used with non-varying pos_to_fragment
> 0(10) : error C7532: global type float requires "#version 100" or later
> 0(10) : error C5060: out can't be used with non-varying select_alpha_to_frag
> 0(12) : error C7532: global type mat4 requires "#version 100" or later
> 0(15) : error C7532: global type mat3 requires "#version 100" or later
> 0(17) : error C7532: global type uint requires "#version 130" or later
> 0(17) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
> 0(32) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
> 0(37) : error C0000: syntax error, unexpected '}' at token "}"
> 0(41) : error C1503: undefined variable "colors"
> 0(41) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '|' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(41) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(46) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(46) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(53) : error C7548: '<<' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(53) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(61) : error C7532: global type int requires "#version 100" or later
> 0(64) : error C7532: global struct false requires "#version 100" or later
> 0(66) : error C7532: global struct true requires "#version 100" or later
> 0(82) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(82) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(82) : error C7011: implicit cast from "uint" to "float"
> 0(83) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(83) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(83) : error C7011: implicit cast from "uint" to "float"
> 0(84) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(84) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(84) : error C7011: implicit cast from "uint" to "float"
> 0(85) : error C7548: '>>' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(85) : error C7548: '&' requires "#extension GL_EXT_gpu_shader4 : enable" before use
> 0(85) : error C7011: implicit cast from "uint" to "float"
> 0(96) : error C7532: global type vec4 requires "#version 100" or later
> 0(99) : error C7532: global variable gl_Position requires "#version 100" or later
> 0(101) : error C7532: global function normalize requires "#version 100" or later
>  
> Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
> 0(2) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
> 0(6) : error C7532: global type vec4 requires "#version 100" or later
> 0(6) : error C5060: out can't be used with non-varying p1_to_geom
> 0(7) : error C5060: out can't be used with non-varying p2_to_geom
> 0(8) : error C7532: global type uint requires "#version 130" or later
> 0(8) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
[... 56 lines cut ...]
> Compile failure in vertex shader: 0(2) : error C0201: unsupported version 330
> 0(2) : error C0000: syntax error, unexpected '(', expecting "::" at token "("
> 0(8) : error C7532: global type vec4 requires "#version 100" or later
> 0(8) : error C5060: out can't be used with non-varying origin_to_geom
> 0(9) : error C7532: global type vec2 requires "#version 100" or later
> 0(9) : error C5060: out can't be used with non-varying shift_to_geom
> 0(10) : error C7532: global type float requires "#version 100" or later
> 0(10) : error C5060: out can't be used with non-varying scale_to_geom
> 0(11) : error C7532: global type uint requires "#version 130" or later
> 0(11) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
> 0(11) : error C5060: out can't be used with non-varying flags_to_geom
> 0(12) : error C5060: out can't be used with non-varying pick_to_geom
> 0(13) : error C5060: out can't be used with non-varying bits_to_geom
> 0(16) : error C7532: global type mat4 requires "#version 100" or later
> 0(20) : error C7532: global variable gl_VertexID requires "#version 130" or later
> 0(20) : error C0000: ... or #extension GL_EXT_gpu_shader4 : enable
> 0(20) : error C7532: global type int requires "#version 100" or later
> 0(21) : error C1503: undefined variable "flags"
> 0(22) : error C1503: undefined variable "origin"
> 0(23) : error C1503: undefined variable "shift"
> 0(24) : error C1503: undefined variable "bits"
> 0(25) : error C1503: undefined variable "scale"
>  
> gl error 1,281 in ../src/canvas/glyph_renderer.cpp:44
> Aborted (core dumped)

wsteffe, 2025-05-08:
The errors were generated running dune3d with Xorg. 
But now I have seen the built dune3d works fine in a Wayland session.

carrotIndustries, 2025-05-08:
Duplicate of #87 

See https://github.com/dune3d/dune3d/issues/87#issuecomment-2096514178 for workaround

closed 2025-05-08 by carrotIndustries