dune3d/dune3d #198: Exception

deviant, 2025-05-08:
Apologies for the nondescript issue title, but I completely forgot exactly what I was doing while in the process of trying to hook up the debugger. I think I may have just deleted a constraint? Or tried to add one? As soon as I dismiss the message box, a new one appears. Here's the backtrace:

```
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
  * frame #0: 0x00007ff8082c71f7 libc++abi.dylib`__cxa_throw
    frame #1: 0x0000000108e097a2 dune3d-bin`std::__1::__throw_out_of_range[abi:fe190107](char const*) + 66
    frame #2: 0x0000000108e24dcb dune3d-bin`std::__1::map<dune3d::UUID, std::__1::unique_ptr<dune3d::Entity, std::__1::default_delete<dune3d::Entity>>, std::__1::less<dune3d::UUID>, std::__1::allocator<std::__1::pair<dune3d::UUID const, std::__1::unique_ptr<dune3d::Entity, std::__1::default_delete<dune3d::Entity>>>>>::at(dune3d::UUID const&) const + 59
    frame #3: 0x0000000108df961d dune3d-bin`dune3d::Entity const& dune3d::Document::get_entity<dune3d::Entity>(dune3d::UUID const&) const + 29
    frame #4: 0x0000000108df95d9 dune3d-bin`dune3d::Document::get_point(dune3d::EntityAndPoint const&) const + 41
    frame #5: 0x0000000109177c9d dune3d-bin`dune3d::Renderer::visit(dune3d::ConstraintPointOnLine const&) + 45
    frame #6: 0x0000000108e8643e dune3d-bin`dune3d::ConstraintPointOnLine::accept(dune3d::ConstraintVisitor&) const + 30
    frame #7: 0x0000000109169d28 dune3d-bin`dune3d::Renderer::render(dune3d::Document const&, dune3d::UUID const&, dune3d::IDocumentView const&, std::__1::__fs::filesystem::path const&, std::__1::optional<dune3d::SelectableRef>) + 3128
    frame #8: 0x000000010905714b dune3d-bin`dune3d::Editor::render_document(dune3d::IDocumentInfo const&) + 1115
    frame #9: 0x0000000109057674 dune3d-bin`dune3d::Editor::canvas_update() + 820
    frame #10: 0x000000010911afec dune3d-bin`dune3d::Editor::tool_process_one() + 92
    frame #11: 0x000000010911abb0 dune3d-bin`dune3d::Editor::tool_process(dune3d::ToolResponse&) + 32
    frame #12: 0x00000001090503ce dune3d-bin`dune3d::Editor::handle_cursor_move() + 126
    frame #13: 0x00000001090730af dune3d-bin`decltype(std::declval<dune3d::Editor&>().*std::declval<void (dune3d::Editor::* const&)()>()()) std::__1::__invoke[abi:fe190107]<void (dune3d::Editor::* const&)(), dune3d::Editor&, void>(void (dune3d::Editor::* const&)(), dune3d::Editor&) + 95
    frame #14: 0x000000010907301d dune3d-bin`std::__1::invoke_result<void (dune3d::Editor::* const&)(), dune3d::Editor&>::type std::__1::invoke[abi:fe190107]<void (dune3d::Editor::* const&)(), dune3d::Editor&>(void (dune3d::Editor::* const&)(), dune3d::Editor&) + 29
    frame #15: 0x0000000109072ff9 dune3d-bin`sigc::bound_mem_functor<void (dune3d::Editor::*)()>::operator()() const + 41
    frame #16: 0x0000000109072fc5 dune3d-bin`decltype(std::declval<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>()()) std::__1::__invoke[abi:fe190107]<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>(sigc::bound_mem_functor<void (dune3d::Editor::*)()>&) + 21
    frame #17: 0x0000000109072fa5 dune3d-bin`std::__1::invoke_result<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>::type std::__1::invoke[abi:fe190107]<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>(sigc::bound_mem_functor<void (dune3d::Editor::*)()>&) + 21
    frame #18: 0x0000000109072f85 dune3d-bin`decltype(auto) sigc::adaptor_functor<sigc::bound_mem_functor<void (dune3d::Editor::*)()>>::operator()<>() const + 21
    frame #19: 0x0000000109072f69 dune3d-bin`sigc::internal::slot_call<sigc::bound_mem_functor<void (dune3d::Editor::*)()>, void>::call_it(sigc::internal::slot_rep*) + 41
    frame #20: 0x0000000108fde76d dune3d-bin`sigc::internal::signal_emit<void, void>::emit(std::__1::shared_ptr<sigc::internal::signal_impl> const&) + 317
    frame #21: 0x0000000108fd8da5 dune3d-bin`sigc::signal_with_accumulator<void, void>::emit() const + 21
    frame #22: 0x00000001091495d5 dune3d-bin`dune3d::Canvas::setup_controllers()::$_7::operator()(double, double) const + 2613
    frame #23: 0x0000000109148b8d dune3d-bin`decltype(std::declval<dune3d::Canvas::setup_controllers()::$_7&>()(std::declval<double const&>(), std::declval<double const&>())) std::__1::__invoke[abi:fe190107]<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>(dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&) + 45
    frame #24: 0x0000000109148b55 dune3d-bin`std::__1::invoke_result<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>::type std::__1::invoke[abi:fe190107]<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>(dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&) + 37
    frame #25: 0x0000000109148b25 dune3d-bin`decltype(auto) sigc::adaptor_functor<dune3d::Canvas::setup_controllers()::$_7>::operator()<double const&, double const&>(double const&, double const&) const + 37
    frame #26: 0x0000000109148af9 dune3d-bin`sigc::internal::slot_call<dune3d::Canvas::setup_controllers()::$_7, void, double, double>::call_it(sigc::internal::slot_rep*, double const&, double const&) + 57
    frame #27: 0x000000010c7177ea libgtkmm-4.0.0.dylib`(anonymous namespace)::EventControllerMotion_signal_motion_connect_callback(_GtkEventControllerMotion*, double, double, void*) + 122
    frame #28: 0x000000010b762bda libgtk-4.1.dylib`_gtk_marshal_VOID__DOUBLE_DOUBLEv + 153
    frame #29: 0x000000010b0f813d libgobject-2.0.0.dylib`_g_closure_invoke_va + 214
    frame #30: 0x000000010b10e049 libgobject-2.0.0.dylib`signal_emit_valist_unlocked + 2575
    frame #31: 0x000000010b10d619 libgobject-2.0.0.dylib`g_signal_emit_valist + 52
    frame #32: 0x000000010b10e534 libgobject-2.0.0.dylib`g_signal_emit + 120
    frame #33: 0x000000010b7d4d5c libgtk-4.1.dylib`gtk_event_controller_motion_handle_event + 68
    frame #34: 0x000000010b7d30af libgtk-4.1.dylib`gtk_event_controller_handle_event + 235
    frame #35: 0x000000010b8ed611 libgtk-4.1.dylib`gtk_widget_run_controllers + 218
    frame #36: 0x000000010b836065 libgtk-4.1.dylib`gtk_propagate_event_internal + 304
    frame #37: 0x000000010b8358cc libgtk-4.1.dylib`gtk_main_do_event + 3039
    frame #38: 0x000000010ba41f14 libgtk-4.1.dylib`_gdk_marshal_BOOLEAN__POINTERv + 122
    frame #39: 0x000000010ba8e504 libgtk-4.1.dylib`gdk_surface_event_marshallerv + 146
    frame #40: 0x000000010b0f813d libgobject-2.0.0.dylib`_g_closure_invoke_va + 214
    frame #41: 0x000000010b10da61 libgobject-2.0.0.dylib`signal_emit_valist_unlocked + 1063
    frame #42: 0x000000010b10d619 libgobject-2.0.0.dylib`g_signal_emit_valist + 52
    frame #43: 0x000000010b10e534 libgobject-2.0.0.dylib`g_signal_emit + 120
    frame #44: 0x000000010ba8d7b0 libgtk-4.1.dylib`gdk_surface_handle_event + 326
    frame #45: 0x000000010ba72cbb libgtk-4.1.dylib`_gdk_event_queue_flush + 55
    frame #46: 0x000000010ba8cfb6 libgtk-4.1.dylib`gdk_surface_flush_events + 30
    frame #47: 0x000000010b0f7f79 libgobject-2.0.0.dylib`g_closure_invoke + 201
    frame #48: 0x000000010b10f033 libgobject-2.0.0.dylib`signal_emit_unlocked_R + 1411
    frame #49: 0x000000010b10dcbd libgobject-2.0.0.dylib`signal_emit_valist_unlocked + 1667
    frame #50: 0x000000010b10d619 libgobject-2.0.0.dylib`g_signal_emit_valist + 52
    frame #51: 0x000000010b10e534 libgobject-2.0.0.dylib`g_signal_emit + 120
    frame #52: 0x000000010ba78188 libgtk-4.1.dylib`gdk_frame_clock_flush_idle + 47
    frame #53: 0x000000010c14ab00 libglib-2.0.0.dylib`g_timeout_dispatch + 65
    frame #54: 0x000000010c14dee9 libglib-2.0.0.dylib`g_main_context_dispatch_unlocked + 262
    frame #55: 0x000000010c14e204 libglib-2.0.0.dylib`g_main_context_iterate_unlocked + 488
    frame #56: 0x000000010c14e260 libglib-2.0.0.dylib`g_main_context_iteration + 55
    frame #57: 0x000000010bff4eb0 libgio-2.0.0.dylib`g_application_run + 528
    frame #58: 0x0000000109016c7e dune3d-bin`main + 62
    frame #59: 0x00007ff807f6a530 dyld`start + 3056
```

deviant, 2025-05-08:
Got another one shortly after reopening Dune 3D. Backtrace is slightly different:

```
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
  * frame #0: 0x00007ff8082c71f7 libc++abi.dylib`__cxa_throw
    frame #1: 0x000000010bc847a2 dune3d-bin`std::__1::__throw_out_of_range[abi:fe190107](char const*) + 66
    frame #2: 0x000000010bc9fdcb dune3d-bin`std::__1::map<dune3d::UUID, std::__1::unique_ptr<dune3d::Entity, std::__1::default_delete<dune3d::Entity>>, std::__1::less<dune3d::UUID>, std::__1::allocator<std::__1::pair<dune3d::UUID const, std::__1::unique_ptr<dune3d::Entity, std::__1::default_delete<dune3d::Entity>>>>>::at(dune3d::UUID const&) const + 59
    frame #3: 0x000000010bc7461d dune3d-bin`dune3d::Entity const& dune3d::Document::get_entity<dune3d::Entity>(dune3d::UUID const&) const + 29
    frame #4: 0x000000010bc745d9 dune3d-bin`dune3d::Document::get_point(dune3d::EntityAndPoint const&) const + 41
    frame #5: 0x000000010bff25ff dune3d-bin`dune3d::Renderer::visit(dune3d::ConstraintPointsCoincident const&) + 79
    frame #6: 0x000000010bcfe1ad dune3d-bin`dune3d::ConstraintT<dune3d::ConstraintPointsCoincident>::accept(dune3d::ConstraintVisitor&) const + 29
    frame #7: 0x000000010bfe4d28 dune3d-bin`dune3d::Renderer::render(dune3d::Document const&, dune3d::UUID const&, dune3d::IDocumentView const&, std::__1::__fs::filesystem::path const&, std::__1::optional<dune3d::SelectableRef>) + 3128
    frame #8: 0x000000010bed214b dune3d-bin`dune3d::Editor::render_document(dune3d::IDocumentInfo const&) + 1115
    frame #9: 0x000000010bed2674 dune3d-bin`dune3d::Editor::canvas_update() + 820
    frame #10: 0x000000010bf95fec dune3d-bin`dune3d::Editor::tool_process_one() + 92
    frame #11: 0x000000010bf95bb0 dune3d-bin`dune3d::Editor::tool_process(dune3d::ToolResponse&) + 32
    frame #12: 0x000000010becb3ce dune3d-bin`dune3d::Editor::handle_cursor_move() + 126
    frame #13: 0x000000010beee0af dune3d-bin`decltype(std::declval<dune3d::Editor&>().*std::declval<void (dune3d::Editor::* const&)()>()()) std::__1::__invoke[abi:fe190107]<void (dune3d::Editor::* const&)(), dune3d::Editor&, void>(void (dune3d::Editor::* const&)(), dune3d::Editor&) + 95
    frame #14: 0x000000010beee01d dune3d-bin`std::__1::invoke_result<void (dune3d::Editor::* const&)(), dune3d::Editor&>::type std::__1::invoke[abi:fe190107]<void (dune3d::Editor::* const&)(), dune3d::Editor&>(void (dune3d::Editor::* const&)(), dune3d::Editor&) + 29
    frame #15: 0x000000010beedff9 dune3d-bin`sigc::bound_mem_functor<void (dune3d::Editor::*)()>::operator()() const + 41
    frame #16: 0x000000010beedfc5 dune3d-bin`decltype(std::declval<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>()()) std::__1::__invoke[abi:fe190107]<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>(sigc::bound_mem_functor<void (dune3d::Editor::*)()>&) + 21
    frame #17: 0x000000010beedfa5 dune3d-bin`std::__1::invoke_result<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>::type std::__1::invoke[abi:fe190107]<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>(sigc::bound_mem_functor<void (dune3d::Editor::*)()>&) + 21
    frame #18: 0x000000010beedf85 dune3d-bin`decltype(auto) sigc::adaptor_functor<sigc::bound_mem_functor<void (dune3d::Editor::*)()>>::operator()<>() const + 21
    frame #19: 0x000000010beedf69 dune3d-bin`sigc::internal::slot_call<sigc::bound_mem_functor<void (dune3d::Editor::*)()>, void>::call_it(sigc::internal::slot_rep*) + 41
    frame #20: 0x000000010be5976d dune3d-bin`sigc::internal::signal_emit<void, void>::emit(std::__1::shared_ptr<sigc::internal::signal_impl> const&) + 317
    frame #21: 0x000000010be53da5 dune3d-bin`sigc::signal_with_accumulator<void, void>::emit() const + 21
    frame #22: 0x000000010bfc45d5 dune3d-bin`dune3d::Canvas::setup_controllers()::$_7::operator()(double, double) const + 2613
    frame #23: 0x000000010bfc3b8d dune3d-bin`decltype(std::declval<dune3d::Canvas::setup_controllers()::$_7&>()(std::declval<double const&>(), std::declval<double const&>())) std::__1::__invoke[abi:fe190107]<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>(dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&) + 45
    frame #24: 0x000000010bfc3b55 dune3d-bin`std::__1::invoke_result<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>::type std::__1::invoke[abi:fe190107]<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>(dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&) + 37
    frame #25: 0x000000010bfc3b25 dune3d-bin`decltype(auto) sigc::adaptor_functor<dune3d::Canvas::setup_controllers()::$_7>::operator()<double const&, double const&>(double const&, double const&) const + 37
    frame #26: 0x000000010bfc3af9 dune3d-bin`sigc::internal::slot_call<dune3d::Canvas::setup_controllers()::$_7, void, double, double>::call_it(sigc::internal::slot_rep*, double const&, double const&) + 57
    frame #27: 0x000000010f5927ea libgtkmm-4.0.0.dylib`(anonymous namespace)::EventControllerMotion_signal_motion_connect_callback(_GtkEventControllerMotion*, double, double, void*) + 122
    frame #28: 0x000000010e5ddbda libgtk-4.1.dylib`_gtk_marshal_VOID__DOUBLE_DOUBLEv + 153
    frame #29: 0x000000010df7313d libgobject-2.0.0.dylib`_g_closure_invoke_va + 214
    frame #30: 0x000000010df89049 libgobject-2.0.0.dylib`signal_emit_valist_unlocked + 2575
    frame #31: 0x000000010df88619 libgobject-2.0.0.dylib`g_signal_emit_valist + 52
[... cut ...]
    frame #54: 0x000000010efc9204 libglib-2.0.0.dylib`g_main_context_iterate_unlocked + 488
    frame #55: 0x000000010efc9260 libglib-2.0.0.dylib`g_main_context_iteration + 55
    frame #56: 0x000000010ee6feb0 libgio-2.0.0.dylib`g_application_run + 528
    frame #57: 0x000000010be91c7e dune3d-bin`main + 62
    frame #58: 0x00007ff807f6a530 dyld`start + 3056
```

Here's what I was drawing:

[image]

The exception occurred after I tried drawing a bezier from the middle point on the bottom edge to the point at the top right corner.

carrotIndustries, 2025-05-08:
Have a file to reproduce this?

deviant, 2025-06-05:
This time, when drawing a segment (no constraints, just freely in the document). Have run into this before, but didn't know how to capture a backtrace when I ran into it then. Pressing ignore makes the popup go away, but as soon as the mouse is moved then a new exception is thrown.

```
* thread #1, queue = 'com.apple.main-thread', stop reason = breakpoint 1.1
  * frame #0: 0x00007ff8118321f7 libc++abi.dylib`__cxa_throw
    frame #1: 0x0000000103e45402 dune3d-bin`std::__1::__throw_out_of_range[abi:fe200100](char const*) + 66
    frame #2: 0x0000000103e5df6b dune3d-bin`std::__1::map<dune3d::UUID, std::__1::unique_ptr<dune3d::Entity, std::__1::default_delete<dune3d::Entity>>, std::__1::less<dune3d::UUID>, std::__1::allocator<std::__1::pair<dune3d::UUID const, std::__1::unique_ptr<dune3d::Entity, std::__1::default_delete<dune3d::Entity>>>>>::at(dune3d::UUID const&) const + 59
    frame #3: 0x0000000103e35ccd dune3d-bin`dune3d::Entity const& dune3d::Document::get_entity<dune3d::Entity>(dune3d::UUID const&) const + 29
    frame #4: 0x0000000103e35c89 dune3d-bin`dune3d::Document::get_point(dune3d::EntityAndPoint const&) const + 41
    frame #5: 0x000000010416016f dune3d-bin`dune3d::Renderer::visit(dune3d::ConstraintPointsCoincident const&) + 79
    frame #6: 0x0000000103eb3c4d dune3d-bin`dune3d::ConstraintT<dune3d::ConstraintPointsCoincident>::accept(dune3d::ConstraintVisitor&) const + 29
    frame #7: 0x00000001041526b8 dune3d-bin`dune3d::Renderer::render(dune3d::Document const&, dune3d::UUID const&, dune3d::IDocumentView const&, std::__1::__fs::filesystem::path const&, std::__1::optional<dune3d::SelectableRef>) + 3128
    frame #8: 0x000000010406458b dune3d-bin`dune3d::Editor::render_document(dune3d::IDocumentInfo const&) + 1115
    frame #9: 0x0000000104064ab4 dune3d-bin`dune3d::Editor::canvas_update() + 820
    frame #10: 0x0000000104108bbc dune3d-bin`dune3d::Editor::tool_process_one() + 92
    frame #11: 0x0000000104108790 dune3d-bin`dune3d::Editor::tool_process(dune3d::ToolResponse&) + 32
    frame #12: 0x000000010405d8ce dune3d-bin`dune3d::Editor::handle_cursor_move() + 126
    frame #13: 0x000000010407ddaf dune3d-bin`decltype(std::declval<dune3d::Editor&>().*std::declval<void (dune3d::Editor::* const&)()>()()) std::__1::__invoke[abi:fe200100]<void (dune3d::Editor::* const&)(), dune3d::Editor&, void>(void (dune3d::Editor::* const&)(), dune3d::Editor&) + 95
    frame #14: 0x000000010407dd1d dune3d-bin`std::__1::invoke_result<void (dune3d::Editor::* const&)(), dune3d::Editor&>::type std::__1::invoke[abi:fe200100]<void (dune3d::Editor::* const&)(), dune3d::Editor&>(void (dune3d::Editor::* const&)(), dune3d::Editor&) + 29
    frame #15: 0x000000010407dcf9 dune3d-bin`sigc::bound_mem_functor<void (dune3d::Editor::*)()>::operator()() const + 41
    frame #16: 0x000000010407dcc5 dune3d-bin`decltype(std::declval<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>()()) std::__1::__invoke[abi:fe200100]<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>(sigc::bound_mem_functor<void (dune3d::Editor::*)()>&) + 21
    frame #17: 0x000000010407dca5 dune3d-bin`std::__1::invoke_result<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>::type std::__1::invoke[abi:fe200100]<sigc::bound_mem_functor<void (dune3d::Editor::*)()>&>(sigc::bound_mem_functor<void (dune3d::Editor::*)()>&) + 21
    frame #18: 0x000000010407dc85 dune3d-bin`decltype(auto) sigc::adaptor_functor<sigc::bound_mem_functor<void (dune3d::Editor::*)()>>::operator()<>() const + 21
    frame #19: 0x000000010407dc69 dune3d-bin`sigc::internal::slot_call<sigc::bound_mem_functor<void (dune3d::Editor::*)()>, void>::call_it(sigc::internal::slot_rep*) + 41
    frame #20: 0x0000000103ff380d dune3d-bin`sigc::internal::signal_emit<void, void>::emit(std::__1::shared_ptr<sigc::internal::signal_impl> const&) + 317
    frame #21: 0x0000000103fee345 dune3d-bin`sigc::signal_with_accumulator<void, void>::emit() const + 21
    frame #22: 0x0000000104133525 dune3d-bin`dune3d::Canvas::setup_controllers()::$_7::operator()(double, double) const + 2613
    frame #23: 0x0000000104132add dune3d-bin`decltype(std::declval<dune3d::Canvas::setup_controllers()::$_7&>()(std::declval<double const&>(), std::declval<double const&>())) std::__1::__invoke[abi:fe200100]<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>(dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&) + 45
    frame #24: 0x0000000104132aa5 dune3d-bin`std::__1::invoke_result<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>::type std::__1::invoke[abi:fe200100]<dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&>(dune3d::Canvas::setup_controllers()::$_7&, double const&, double const&) + 37
    frame #25: 0x0000000104132a75 dune3d-bin`decltype(auto) sigc::adaptor_functor<dune3d::Canvas::setup_controllers()::$_7>::operator()<double const&, double const&>(double const&, double const&) const + 37
    frame #26: 0x0000000104132a49 dune3d-bin`sigc::internal::slot_call<dune3d::Canvas::setup_controllers()::$_7, void, double, double>::call_it(sigc::internal::slot_rep*, double const&, double const&) + 57
    frame #27: 0x00000001073417ea libgtkmm-4.0.0.dylib`(anonymous namespace)::EventControllerMotion_signal_motion_connect_callback(_GtkEventControllerMotion*, double, double, void*) + 122
    frame #28: 0x000000010638a7b6 libgtk-4.1.dylib`_gtk_marshal_VOID__DOUBLE_DOUBLEv + 153
[... cut ...]
    frame #53: 0x0000000106d74aac libglib-2.0.0.dylib`g_timeout_dispatch + 65
    frame #54: 0x0000000106d77e95 libglib-2.0.0.dylib`g_main_context_dispatch_unlocked + 262
    frame #55: 0x0000000106d781b0 libglib-2.0.0.dylib`g_main_context_iterate_unlocked + 488
    frame #56: 0x0000000106d7820c libglib-2.0.0.dylib`g_main_context_iteration + 55
    frame #57: 0x0000000106c1eeb0 libgio-2.0.0.dylib`g_application_run + 528
    frame #58: 0x0000000104028d8e dune3d-bin`main + 62
    frame #59: 0x00007ff8114d4530 dyld`start + 3056
```

carrotIndustries, 2025-06-05:
with 82fba8f, the renderer catches and logs the exception so that the app stays somewhat usable

carrotIndustries, 2025-06-05:
Critical log items will automatically open the logger window.