dune3d/dune3d #203: Clipping the view to the active sketch workplane

valpackett, 2025-05-18:
Say I wanna sketch on a workplane aligned to the inside of a box:

[image]

Well, if I actually align with the workplane, I can't see anything I'm sketching as it's occluded by the front side of the box:

[image]

Unless I manually turn on a clipping plane:

[image]

In this simple test case it's enough to have it at 0, but in general it's very useful to clip the view exactly to the sketch workplane. FreeCAD always defaults to that when entering the sketcher and even makes it not-so-obvious how to enter the full view (there's a button).

carrotIndustries, 2025-05-18:
I think clipping by default like in FreeCAD won't work for Dune 3D since we don't have a modal sketcher.

I usually work around this by hiding the solid model of the current body.

A possible enhancement could be adding another toggle to the clipping setting for ciipping the solid model on either side of the active workplane.

Clipping itself could also use some refinement so that the solid model doesn't appear hollow.

valpackett, 2025-05-18:
Yeah, the first thing would be to just have a button / tool / toggle / thingy for aligning the clipping plane to the workplane.

I'd like it to be optionally semi-"modal" though, with a setting that would trigger it automatically upon selecting a sketch in the tree (and readjust it if the workplane changes) and disable the clipping upon selecting something else.

> Clipping itself could also use some refinement so that the solid model doesn't appear hollow

yep, like realthunder's fork of FreeCAD that has nice hatch patterns for that. Also a visual manipulator for the clipping plane.

carrotIndustries, 2025-05-20:
The thing is, the rendering of the clipped solid model leaves a lot to be desired right now. I had a go at improving it somewhat by rendering the inside faces in a solid color without shading, but couldn't get normals that are the consistently right way round out of opencascade.

Making clipping planes a more prominent feature will inevitably lead to users complaining about the poor rendering.

valpackett, 2025-05-20:
I might get around to looking into how to implement the hatch thing…

carrotIndustries, 2025-05-21:
Had some more thoughts about what could be done if we can't get the normals facing the right way round: It mostly boils down to actually rendering the flat cap to close out the now-open solid model.

 - Slice the triangulated representation across the clipping plane and try to stitch the resulting lines to contours and turn them into polygons using the even/odd filling rule.
 - The same thing, but on the OpenCascade representation using BRepAlgoAPI_Section

carrotIndustries, 2025-05-21:
I think working on the triangulated representation is the way to go as I prefer keeping opencascade usage to a minimum.
One thing to consider is that we should add `solid_id` member to the `Face` class so that we'll only make contours out of edges from single solid. Contours from a single solid are to be combined using even/odd filling rule, but contours from different solids should be combined by unioning them.