dune3d/dune3d #205: Frequent crashes on wayland

ii8, 2025-06-07:
When used with specific wayland compositors Dune3D will often crash with the message:
```
Gdk-Message: 15:14:20.989: Lost connection to Wayland compositor.
```
See [here](https://gitlab.gnome.org/GNOME/gtk/-/blob/main/gdk/wayland/gdkeventsource.c)

This issue is related to the following conundrum in wayland: https://gitlab.freedesktop.org/wayland/wayland/-/issues/159

It happens with Dune3D specifically whenever something is recomputed by (I'm guessing) occt and the main GTK loop is blocked and therefore not reading events from the wayland fd.

To reproduce:

- Do something that makes Dune3D freeze for a couple of seconds while it recomputes.
- Move the mouse pointer around while over the dune window. This will flood the file descriptor with events that GTK doesn't read and cause the overflow.

Note that some wayland compositors, notably GNOME, have workarounds for this issue, they stop sending pointer motion events when a client is not responding. But this crash happens on at least wlroots and I think KDE plasma Kwin as well.

It is possible to workaround the problem by running dune3d with https://github.com/stransky/wayland-proxy
This works well for now and this is actually what Firefox does to fix this same issue, but it would probably make more sense for dune3d to run operations that can block the GTK event loop in a separate thread so that it always remains responsive.

carrotIndustries, 2025-06-09:
Any idea whether the majority of the time is spent in solid model update or solving the sketches?

ii8, 2025-06-09:
Looks like it happens while solving, during the time where a bunch of `solve group ...` messages are getting printed to the console.

[test.zip](https://github.com/user-attachments/files/20657594/test.zip)

If you unzip and open this in dune, then go to the beginning of the body, for example click on the "Reference" group, and then add a new sketch there: it resolves everything below and if you move your mouse around during this time it crashes. If you don't move the mouse then you can add a rectangle to the sketch and again it does solving and moving the mouse here too crashes. You can resize the rectangle, extrude it, etc. each time all groups are re-solved you get a chance to crash if you accidentally move the mouse. If your PC is super fast and resolves all the groups instantly it might not happen for you.

carrotIndustries, 2025-06-09:
Ok, try 87c5a099a5024bc0dc1e6ebe0cee096ca5ac6196. This prints the time it takes to update the solid model.

For me, the longest is Fillet 1 with about 1.1 seconds.

The solving itself is always in the millisecond range.

carrotIndustries, 2025-06-09:
Ok, this confirms my hunch that the solid model update is indeed the culprit here. What follows is a braindump of how we could implement asynchronous solid model updates:

A key thing to keep in mind is that the the background thread can't access the document as that could change at any time. To remove this dependency, the solid model update is split into two steps:
 1. Collecting everything required from the document, such as building paths from the source entities, body color, previous solid model group, etc. This will still happen in the UI thread as-is, but shouldn't be a problem since it's not the expensive part. As this can be done individually for every group once they're all solved, this could be done in parallel for all groups.
 2. Building the actual solid model can then be done in a background thread without access to the document based on the information collected in step 1.

While we're at it, building the solid model can be massively parallelized as well by also breaking it up into two steps:
 1. Creating the local solid model, for example an extrusion or lathe
 2. Creating the accumulated solid model by combining the local solid model with the one from the previous group using the group's solid model operation (union/difference/intersection)

The first step can be done in parallel for all groups. Only the second step has to be done sequentially per body.

The `SolidModel` class will mostly stay as-is, that is it'll be recreated as part of `Document::update_pending`, but will remain in an empty state with everything collected from the document. To still have something to draw, the new solid model retains a reference to the old one.

To populate the empty solid models, the `update_pending` method submits all newly-created solid models to a central per-document dispatcher that distributes the individual tasks (local/accumulated) to number of worker threads. The dispatcher then informs the UI about the state of the solid model update.

Due to the asynchronous nature of the solid model update, a new `update_pending` could be triggered, while the solid models form the previous run are still being generated. The dispatcher will then handle this by replacing queued groups with the new ones. Depending on the exact timing, the new solid model update could depend on solid model from the previous run that haven't yet been generated, but this should be no problem for the dispatcher to handle.

ii8, 2025-06-10:
Another question to consider would be: can the user continue working while the solid model updates happen in the background? In theory I would assume yes because only the current group needs to be updated for the user to get an up-to-date view of what is actually being worked on. The groups below don't matter.

If not, the wayland proxy method might just be a lot simpler and the result the same.

ii8, 2025-06-10:
Might want to look here for the proxy code: https://github.com/mozilla-firefox/firefox/tree/main/third_party/wayland-proxy
Looks like the code at https://github.com/stransky/wayland-proxy is out of date by a bit