dune3d/dune3d #208: Loft group does not support some boundaries.

drwho495, 2025-06-28:
Hello, there! Dune3D's loft group is missing boundary generation (those lines that the user selects in something like an extrusion to create constraints) for edges not directly created by sketches. There is a way to implement these; I was able to do it in my FreeCAD addon, which implements boundaries along with a design workbench.

Here is a picture of what the boundaries would look like:
[image]

The method that I use to find those highlighted edges goes like this:
1. Loop through all the edges generated by opencascade
2. Check to make sure they are not in the same plane has the sketches (they aren't apart of the sketch)
3. Check all the vertexes used to create the line, see if any of them are created by a sketch
4. If any of them are, then store the geometry IDs of the sketch geometry which created those vertexes
5. Add this edge to the TopoShape

Here is a look at the python:
[image]

drwho495, 2025-06-28:
also if you want to support inner geometry for a loft, you can do two operations and cut the inner geometry with the outer geometry

carrotIndustries, 2025-06-29:
The problem with using anything generated from OpenCascade is the topological naming problem

> Add this edge to the TopoShape

That's not how Dune 3D works. OpenCascade and its data structures are only used for the solid model.

drwho495, 2025-06-29:
> The problem with using anything generated from OpenCascade is the topological naming problem
> 
> > Add this edge to the TopoShape
> 
> That's not how Dune 3D works. OpenCascade and its data structures are only used for the solid model.

That's why you would filter through the open Cascade edges each time to apply a persistent name that would mitigate the problem topological naming problem

drwho495, 2025-07-01:
> > That's why you would filter through the open Cascade edges each time to apply a persistent name 
> 
> Ok, so how do I do this? That could also come in handy to fix the topological naming problem for fillets and chamfers.
> 
> This is the place where Opencascade edges are created from entities: https://github.com/dune3d/dune3d/blob/main/src/document/solid_model/solid_model_util.cpp#L379

Well, in the original post I described the method I used (and sent a screenshot of python code) in my Freecad addon. It is very stable from the testing I've done. Fixing fillets is easier than this, what I did for those is have the user select the boundaries (not the open cascade elements) and I do a check to see if any of the opencascade elements intersect the selected boundaries, and if it does, then just fillet the occt element like normal.