dune3d/dune3d #219: Segmentation fault in STEP importer

Milliam21, 2025-08-06:
This bug was observed on the following commit hash: b8a24a6274ce3b0f9665fd69b880d7b3c017141d

Bug: Segmentation fault in STEP importer

- Summary: The application crashes due to a segmentation fault when a user attempts to import a specially crafted .step file. This appears to be an application-level denial-of-service bug.
- Component: STEP file importer and its interaction with the OpenCASCADE library (libTKernel.so.7).
- Mechanism: An attempt to import a specially crafted .step file causes an unhandled error during parsing within the OpenCASCADE backend. The program's output indicates a syntax error is detected, but it fails to recover gracefully.
- Result: The unhandled error leads to a segmentation fault (SIGSEGV), which immediately crashes the entire Dune3D application.

Steps to Reproduce:
1. Launch Dune3D.
2. Attempt to import the proof of concept file, test-case.step (included in this issue).
3. The application will crash.

Note: The proof-of-concept file has been uploaded as test-case.txt because GitHub does not allow .step file attachments. Please rename it to test-case.step to trigger the bug.

[test-case.txt](https://github.com/user-attachments/files/21633110/test-case.txt)

carrotIndustries, 2025-08-06:
I'm a bit hesitant to open a specially-crafted STEP file causing a segfault on my machine, especially from a user who just opened their first issue in this project and joined github this month.

Have a backtrace?

Does this reproduce in FreeCAD or other applications importing STEP with opencascade?

carrotIndustries, 2025-08-06:
Ok, so the bug is entirely within OpenCascade and should be reproducible using a test case consisting of these two lines:
```c++
    STEPCAFControl_Reader reader;
    IFSelect_ReturnStatus stat = reader.ReadFile(fname);
```

How did you come up with this test case and how did it end up containing mangled symbol names from the solvespace solver? Some kind of fuzzer?

Milliam21, 2025-08-07:
These inputs were found via RAM Lab's Shorebreak tool version 1.3 which is an AI powered fuzzer-based application for finding bugs and vulnerabilities. We are currently in the process of testing this on various open-source projects including Dune3D, and discovered this bug. After looking into it briefly we do not believe it is an exploitable vulnerability, but we still wanted to do our due diligence and report our findings. I'm happy to help in any way possible and provide additional information if requested!

Milliam21, 2025-08-07:
@ruevs I would be happy to test SolveSpace! I'll add it to the queue and keep you updated!