dune3d/dune3d #226: App can't launch

chrostino, 2025-09-19:
app doesn't launch at all, it doesn't even display a window before crashing. I'm on Fedora 42, wayland with Nvidia graphic card with driver version 580.82.09. App was installed via flatpak

`flatpak run org.dune3d.dune3d` results in:
```
pci id for fd 18: 10de:2489, driver (null)
kmsro: driver missing
realize
dune3d: ../src/dispatch_common.c:872: epoxy_get_proc_address: Assertion `0 && "Couldn't find current GLX or EGL context.\n"' failed.
```

carrotIndustries, 2025-09-19:
Do the OpenGL demos from `gtk4-demo` work?
Did you try building and running locally?

carrotIndustries, 2025-09-19:
Try setting the environment variable `GDK_DEBUG=gl-prefer-gl` and see if it makes a difference.

What about

>  building and running locally to rule out that it's a flatpak issue?